AlexVerrico/Octoprint-ThermalRunaway #3: OctoPi says the plugin has invalid python

qbm5, 2020-06-29:
Plugin does not work for me (OctoPrint 1.4.0 running on OctoPi 0.17.0)  Get a lightning bolt in the plugin manager that says the plugin is invalid.

AlexVerrico, 2020-06-29:
Hi @qbm5
What version of python are you using?
Cheers,
Alex.

qbm5, 2020-06-29:
Full disclosure… I am not too familiar with the way Linux and octopi work under the hood, all I know is the installed version
·         OctoPrint 1.4.0 running on OctoPi 0.17.0
I tried to install via the plugin manager and the search did not return your plugin

[cid:image001.png@01D64E3E.CEB95680]

So I installed it manually via the url.  After I activate the plugin it says this.
[cid:image004.png@01D64E3F.66E506C0]

AlexVerrico, 2020-06-29:
Hi @qbm5 
Can you please try installing the plugin from this link: https://github.com/AlexVerrico/Octoprint-ThermalRunaway/archive/Python2-test.zip
Cheers,
Alex

qbm5, 2020-06-29:
I have a large print going, I will try it out when I can.  Thank you.

vituslehner, 2020-07-01:
I'm having the same issue with installing the plugin (not able to install & enable via Plugin Manager or manual link). Using your alternative installation link, however, works fine. Am using OctoPrint 1.4.0 too, but indeed my OctoPi setup seems to be running on Python 2.7.

AlexVerrico, 2020-07-02:
Hi @qbm5 and @vituslehner 
Sorry for not responding sooner.
The reason why the plugin doesn't show in the repository for you is that I marked it as only being Python 3 compatible as I can't test on Python 2 currently. Since @vituslehner has said that the alternative link above worked, I will now mark the main link as Python 3 compatible.
Cheers,
Alex.

closed 2020-07-02 by AlexVerrico

AlexVerrico, 2020-07-02:
You should now be able to install this plugin through the plugin manager.