AlexVerrico/Octoprint-ThermalRunaway #5: Connecting to a printer that is already hot results in immediate emergency kill

fionnb, 2020-07-05:
If I have this plugin enabled and reconnect octoprint to a printer that is already hot, the plugin triggers an emergency kill after less than 2 seconds. I have to wait for the printer to completely cool down or disable the plugin to be able to reconnect the printer.

AlexVerrico, 2020-07-05:
Hi @fionnb 
Can you try changing the delay before triggering a thermal runaway (the bottom 2 fields in the settings page) to 30 seconds for both the hotend and bed?

AlexVerrico, 2020-07-15:
Hi @fionnb 
Did you try changing the values that I suggested?
Cheers,
Alex.

AlexVerrico, 2020-07-21:
Hi @fionnb.
This issue will be closed in 1 week if there is no further activity.
Cheers,
Alex

AlexVerrico, 2020-08-02:
Closing due to no activity.

closed 2020-08-02 by AlexVerrico