AlexVerrico/Octoprint-ThermalRunaway #6: Cannot find plug-in settings.

thanarious, 2020-07-05:
Maybe it's me...
Installed the plug-in, refreshed the octoprint interface as instructed, but I can't seem to be able to find the plug-in settings to set the proper temp and time limits anywhere. Don't see anything regarding the issue on the readme, either.
Thank for any help.

thanarious, 2020-07-05:
OK, a restart of the server was needed, seems you'll have to properly update the post-install script then...

AlexVerrico, 2020-07-05:
Octoprint always requests a restart when I install this plugin, so I'm not sure what went wrong.

thanarious, 2020-07-05:
It only asked for a browser refresh on my case, hence the initial question.

AlexVerrico, 2020-07-05:
Ok then, I will look into it at some point.
Cheers,
Alex.

merrill-oakland, 2021-03-02:
This is still an issue. I just installed the plugin for the first time, and it asked me to reload instead of reboot. Had to reboot manually to get the plugin to show up. Only plugin I've seen this with.
I think your plugin might not be using a hook or mixin that normally requires a restart, but it does indeed need a restart to work (seemingly). 
I _think_ adding octoprint.plugin.RestartNeedingPlugin to https://github.com/AlexVerrico/Octoprint-ThermalRunaway/blob/master/octoprint_ThermalRunaway/__init__.py#L12 would fix it

AlexVerrico, 2021-03-03:
Hmm, I'll have to take another look at this.

StephanBeutel, 2021-03-07:
I had the same problem - deleting cookies solved the problem.

AlexVerrico, 2021-03-24:
G'Day @merrill-oakland @StephanBeutel 
I believe that I have fixed this in the Dev branch now, thanks again for reporting the issue.
Cheers, Alex.

commit 4c5d6e14 (2021-03-24): Fix #6 and bump version to 2.0.1
commit 867c9be6 (2021-03-29): Fix #6  (#19)
closed 2021-03-24 by AlexVerrico