AlexVerrico/Octoprint-ThermalRunaway #9: [BUG] text unclear

njilk, 2020-07-15:
**Describe the bug**
for the text: 
Maximum amount that the bed / hotend temp is allowed **to differ**  from the set temp before triggering a Thermal Runaway:

I assume you mean to say higher in stead of differ? Because lower is obvious  when initiating a print and it wouldn't be a thermal runaway.
So than the text should be: "is allowed to be higher than the set temperature" 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the plugin
2. read the text:
Maximum amount that the bed / hotend temp is allowed **to differ**  from the set temp before triggering a Thermal Runaway:

**Expected behavior**
So than the text should be: "is allowed to be higher than the set temperature" 

Sorry for nitpicking ;-)

AlexVerrico, 2020-07-15:
Hi @njilk 
I meant to say **differ**, as the plugin is checking for a thermal runaway, which can be under temp or over temp.
Cheers,
Alex.

closed 2020-07-15 by AlexVerrico

AlexVerrico, 2020-07-15:
@njilk I should also mention, if the temperature is moving towards the set temp (like when the printer is heating up), a thermal runaway won't trigger.

njilk, 2020-07-16:
Hello Alex,

According to wikipedia:
https://en.wikipedia.org/wiki/Thermal_runaway#:~:text=Thermal%20runaway%20occurs%20in%20situations,kind%20of%20uncontrolled%20positive%20feedback.

"In other words, "thermal runaway" describes a process which is accelerated by increased temperature, in turn releasing energy that further increases temperature."

It is all about an uncontrolled increase of temperature. Under temperature or decrease, is not related to thermal runaway.
I understand why you say differ, so maybe the plug in name is not covering its full functionality.

Regards,
Ad

AlexVerrico, 2020-07-16:
Hi @njilk 
My plugin is named after the Marlin 3D Printer firmware thermal runaway protection, which only checks for under temp. I wasn't aware that it is technically incorrect, but I still think it is the best name as a lot of people with 3d printers will know what it means.
The purpose of "Thermal Runaway" protection on a 3d printer is to detect if a heater fails and either gets stuck on or fails to turn on, and doesn't necessarily align with the technical definition.
Cheers,
Alex.

AlexVerrico, 2020-07-16:
I don't mean to come across as rude, but if you really have a problem with this term been used incorrectly, you would be better to take it up over at the Marlin Firmware repo.

njilk, 2020-07-16:
Well, I don’t mind a discussion as long as the intention is positive.

My point is that an under temperature, so the hot end has a lower temperature than the setting or specified temperature, is not a thermal runaway. If it is a higher one, then yes it is.
So a lower than specified temperature can be a real problem for the print, but it is not a thermal runaway…….

This is what Marlin says about it: https://marlinfw.org/docs/configuration/configuration.html
"Monitor thermal stability.
If the measured temperature drifts too far from the target temperature for too long, the machine will shut down with a “Thermal runaway” error. This error may indicate poor contact between thermistor and hot end, poor PID tuning, or a cold environment."

So yes you are right, Marlin is also abusing the therm. I had not realised that.

AlexVerrico, 2020-07-16:
Thanks for the suggestion, but I think that I will keep the current name for consistency with the rest of the 3d printing community/software.
Cheers,
Alex.