AlexVerrico/Octoprint-ThermalRunaway #10: [BUG]

bjenkins2009, 2020-08-01:
**Describe the bug**
On hot summer days without AC, the room with my printer can exceed 30 degrees C without issue. Octopi then disconnects shortly after connecting and I see some variation of:
`Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Closing serial port due to emergency stop M112.`
Where the first and final messages are always the same

**To Reproduce**
Steps to reproduce the behavior:
1. Put printer in a room > 30C and let the plate and hotend reach this temperature
2. Click connect in octopi with default thermal-runaway settings
3. Wait (if M112 isnt sent at this point, load a job and press preheat)
4. At this point I have always had Octopi send M112 and disconnect

**Expected behavior**
I expect no issues with octopi and printing, or a message to indicate that this add-on is what sent the emergency stop command.

**Screenshots**
None

**Environment information:**
 - Octoprint Version 1.4.0
 - Octopi Version 0.17.0, running on Raspberry Pi 3 Model B Rev 1.2
 - Browser: Chrome Version 84.0.4147.105 (Official Build) (64-bit)
 - Python Version (whatever came with octopi)

**Additional context**
I understand that increasing the allowable temperatures in the thermal-runaway add-on will fix this, but there is not any message in the terminal indicating that it was the thermal-runaway add-on that was causing the issue. Can there at least be a message in terminal indicating the reason for the emergency stop? I've had everything running for quite some time without any issues so it was hard to pinpoint why suddenly the system was no longer working.

AlexVerrico, 2020-08-02:
Hi @bjenkins2009
There is an option to configure the maximum temperature while turned off for both the hotend and bed.
In regards to the error message, thanks for the idea, I will look into this for inclusion in the next release.
Cheers,
Alex.

AlexVerrico, 2020-08-02:
Since this issue is technically invalid, I will close this issue, but will also open a feature request for the error message.

closed 2020-08-02 by AlexVerrico