AlexVerrico/Octoprint-ThermalRunaway #13: [BUG]

ssepeq, 2020-09-02:
**Describe the bug**
Sends M112 kill signal when first connecting to the printer, if the nozzle is already hot before the printer was reset. Sometimes this happens even if the nozzle was not previously hot.

**To Reproduce**
Steps to reproduce the behavior:
1. Heat the nozzle
2. Reset the printer, then reconnect to it (bug sometimes happens without this step)
3. Begin a print, or set the hotend to temp (bug sometimes happens without this step)
4. Look in terminal tab. Plugin sends M112 causing the printer to shut off.

**Expected behavior**
Should be able to heat the hotend and nozzle and print things even if a previous attempted print failed.

**Environment information:**
 - Octoprint Version: 1.4.2
 - Browser: Chrome
 - Python Version: 2.7.16 or 3.7.3, whichever it uses by default
 - Printer: Creality Ender 5 Pro

**Additional context**
Temperature is always decreasing when connecting to a machine that was previously heated; the code should detect this when recently connected and not trigger a runaway. There is a setting called "Maximum temp that the hotend is allowed to reach while turned off before triggering a Thermal Runaway" but surely it's not reasonable to set that to 200C? This bug never occurs when heating the bed, only when heating the nozzle.

AlexVerrico, 2020-09-02:
Hi Julie.
Can you please try increasing the "maximum time that the nozzle can be outside of the set temp" to 30 seconds, and if it still errors out then try 60 seconds?
If that doesn't fix it, let me know.
Cheers,
Alex

ssepeq, 2020-09-05:
Didn't fix it.

AlexVerrico, 2020-09-05:
Hi @ssepeq 
I was unable to reproduce this issue on Octoprint 1.4.0, and won't be able to upgrade to 1.4.2 for a few weeks.
Once i have upgraded, I will revisit this bug.
In the meantime, PRs are always welcome 😉 
Cheers,
Alex.

AlexVerrico, 2020-10-07:
Hi @ssepeq 
I believe that I have fixed this issue in the latest release, if you continue to have trouble with this please let me know.
If I don't hear back within a week then I will consider this issue resolved.
Cheers,
Alex.

AlexVerrico, 2020-10-14:
Closing due to inactivity

PR #15 (2020-10-07): Version2
Version 2.0.0
Mainly a major code rewrite to use a more pythonic approach, including using loops rather than copy and pasting code.
Also fixes #13
commit fa60b9bc (2020-10-07): Merge pull request #15 from AlexVerrico/version2

Version 2.0.0
Mainly a major code rewrite to use a more pythonic approach, including using loops rather than copy and pasting code.
Also fixes #13
PR #16 (2020-10-07): Version 2
Version 2.0.0
Mainly a major code rewrite to use a more pythonic approach, including using loops rather than copy and pasting code.
Also fixes #13
Also includes a minor code cleanup including adding more comments to make it easier for others to contribute.
commit b60986f4 (2020-10-07): Version 2 (#16)

Version 2.0.0
Mainly a major code rewrite to use a more pythonic approach, including using loops rather than copy and pasting code.
Also fixes #13
Also includes a minor code cleanup including adding more comments to make it easier for others to contribute.
closed 2020-10-14 by AlexVerrico