AlexVerrico/Octoprint-ThermalRunaway #17: [BUG] Throwing Python exception into the docker logs

TheFlyingBadger, 2021-01-21:
**Describe the bug**
TypeError: not all arguments converted during string formatting

**To Reproduce**
Steps to reproduce the behavior:
Launch the docker container...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment information:**
 - Octoprint Version: 1.5.3
 - Browser: n/a
 - Python Version: 3.8.7
 - Operating system that Octoprint is running on: Latest Docker Container (created 2021-01-19T10:37:02.068327827Z)

**Additional context**
```
octoprint    | 2021-01-21T11:21:59.852656992Z Exception in thread Thread-1919:
octoprint    | 2021-01-21T11:21:59.852689509Z Traceback (most recent call last):
octoprint    | 2021-01-21T11:21:59.852706312Z   File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
octoprint    | 2021-01-21T11:21:59.852713219Z     self.run()
octoprint    | 2021-01-21T11:21:59.852726535Z   File "/usr/local/lib/python3.8/site-packages/sentry_sdk/integrations/threading.py", line 69, in run
octoprint    | 2021-01-21T11:21:59.852733659Z     reraise(*_capture_exception())
octoprint    | 2021-01-21T11:21:59.852739667Z   File "/usr/local/lib/python3.8/site-packages/sentry_sdk/_compat.py", line 54, in reraise
octoprint    | 2021-01-21T11:21:59.852746278Z     raise value
octoprint    | 2021-01-21T11:21:59.852763286Z   File "/usr/local/lib/python3.8/site-packages/sentry_sdk/integrations/threading.py", line 67, in run
octoprint    | 2021-01-21T11:21:59.852769821Z     return old_run_func(self, *a, **kw)
octoprint    | 2021-01-21T11:21:59.852775401Z   File "/usr/local/lib/python3.8/threading.py", line 1254, in run
octoprint    | 2021-01-21T11:21:59.852781282Z     self.function(*self.args, **self.kwargs)
octoprint    | 2021-01-21T11:21:59.852787031Z   File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_ThermalRunaway/__init__.py", line 130, in check_temps
octoprint    | 2021-01-21T11:21:59.852792662Z     _logger.debug('bTemps: %s' % bTemps)
octoprint    | 2021-01-21T11:21:59.852798340Z TypeError: not all arguments converted during string formatting
```

AlexVerrico, 2021-01-21:
G'day @TheFlyingBadger 
Does your printer have a heated bed?
Cheers, Alex.

TheFlyingBadger, 2021-01-21:
Yep. It's a pxmalion coreI3

TheFlyingBadger, 2021-01-28:
A modified init.py which doesn't crash for me...

```
# coding=utf-8
from __future__ import absolute_import

import octoprint.plugin  # Import the core octoprint plugin components
import logging           # Import logging to allow for easier debugging
import threading         # Import threading. This is used to process the temps asynchronously so that we don't block octoprints communications with the printer
import time              # Import time. This is so that we can add delays to parts of the code

_logger = logging.getLogger('octoprint.plugins.ThermalRunaway')

heaterDict = dict(B     = dict  (highTemp           = 0.0
                                ,lowTemp            = 0.0
                                ,thermalHighWarning = False
                                ,thermalLowWarning  = False
                                ,thermalHighAlert   = False
                                ,thermalLowAlert    = False
                                )
                 ,T0    = dict  (highTemp           = 0.0
                                ,lowTemp            = 0.0
                                ,thermalHighWarning = False
                                ,thermalLowWarning  = False
                                ,thermalHighAlert   = False
                                ,thermalLowAlert    = False
                                )
                 )

class ThermalRunawayPlugin(octoprint.plugin.StartupPlugin,
                           octoprint.plugin.SettingsPlugin,
                           octoprint.plugin.TemplatePlugin):
    def on_after_startup(self):

        # global heaterList
        # global heaterDict
        # heaterDict = dict()
        # heaterDict['B'] = {'highTemp': 0.0, 'lowTemp': 0.0,
        #                    'thermalHighWarning': False, 'thermalLowWarning': False,
        #                    'thermalHighAlert': False, 'thermalLowAlert': False}

        # heaterDict['T0'] = {'highTemp': 0.0, 'lowTemp': 0.0,
        #                     'thermalHighWarning': False, 'thermalLowWarning': False,
        #                     'thermalHighAlert': False, 'thermalLowAlert': False}

#        # create global variables for storing bed temps and thermal warnings
#        global bHighTemp
#        global bLowTemp
#        global bThermalHighWarning
#        global bThermalHighAlert
#        global bThermalLowWarning
#        global bThermalLowAlert
#
#        # create global variables for storing tool temps and thermal warnings
#        global tHighTemp
#        global tLowTemp
#        global tThermalHighWarning
#        global tThermalHighAlert
#        global tThermalLowWarning
#        global tThermalLowAlert

#        # set initial values of bed variables
#        bHighTemp = 0.0
#        bLowTemp = 0.0
#        bThermalHighWarning = False
#        bThermalHighAlert = False
#        bThermalLowWarning = False
#        bThermalLowAlert = False
#
#        # set initial values of tool variables
[... 282 lines cut ...]
                displayVersion=self._plugin_version,

                # version check: github repository
                type="github_release",
                user="AlexVerrico",
                repo="Octoprint-ThermalRunaway",
                current=self._plugin_version,

                # update method: pip
                pip="https://github.com/AlexVerrico/Octoprint-ThermalRunaway/archive/{target_version}.zip"
            )
        )

__plugin_pythoncompat__ = ">=2.7,<4"  # python 2 and 3

def __plugin_load__():
    global __plugin_implementation__
    __plugin_implementation__ = ThermalRunawayPlugin()

    global __plugin_hooks__
    __plugin_hooks__ = {
        "octoprint.plugin.softwareupdate.check_config": __plugin_implementation__.get_update_information,
        "octoprint.comm.protocol.temperatures.received": __plugin_implementation__.get_temps
    }
```

AlexVerrico, 2021-03-24:
G'Day @TheFlyingBadger 
Sorry for not responding for so long, I'm happy to look into this issue further, but it would be easiest if you could fork the repo and make your changes to your fork so that I can easily see the difference.
Cheers, Alex.

zero-impedance, 2021-05-05:
Hi, @AlexVerrico

I also got the same error and faced another problem.

### Environment information:

- Octoprint Version: 1.5.3
- Printer: Anet A8
- Firmware: Repetier
- Python Version: 3.9.4
- Linux raspberrypi 5.4.83+ #1379 Mon Dec 14 13:06:05 GMT 2020 armv6l GNU/Linux

### `temps` data cleared

I continuously got this error message on printing.

~~~
Exception in thread Thread-33:
Traceback (most recent call last):
  File "/home/octoprint/.pyenv/versions/3.9.4/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/home/octoprint/.pyenv/versions/3.9.4/lib/python3.9/threading.py", line 1266, in run
    self.function(*self.args, **self.kwargs)
  File "/home/octoprint/.pyenv/versions/3.9.4/envs/octoprint/lib/python3.9/site-packages/octoprint_ThermalRunaway/__init__.py", line 130, in check_temps
    bTemps = temps['B']
KeyError: 'B'
~~~

After a bit of research, I found that at the beginning of `check_temps()`, argument `temps` had correct data (like `{'B': (25.47, 50.0), 'T0': (27.79, 100.0)}`).  But just before `bTemps = temps['B']` (line 130), `temps` became empty dict (`{}`).

I think the cause may be that OctoPrint cleared `parsed_temps` dict just after `octoprint.comm.protocol.temperatures.received` callback (`get_temps()`), and before using `temps` in `check_temps()`.

Using a copy of `parsed_temps` worked for me.

~~~
     def get_temps(self, comm, parsed_temps):
         _logger.debug('Temps received')  # Log that temps have been received
+        temps = parsed_temps.copy()
-        temps = parsed_temps
         if temps == parsed_temps:
~~~

### Logging format error

Logging `bTemps` and `tTemps` produced an error as @TheFlyingBadger reported.

~~~
Exception in thread Thread-31:h
Traceback (most recent call last):
  File "/home/octoprint/.pyenv/versions/3.9.4/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/home/octoprint/.pyenv/versions/3.9.4/lib/python3.9/threading.py", line 1266, in run
    self.function(*self.args, **self.kwargs)
  File "/home/octoprint/.pyenv/versions/3.9.4/envs/octoprint/lib/python3.9/site-packages/octoprint_ThermalRunaway/__init__.py", line 131, in check_temps
    _logger.debug('bTemps: %s' % bTemps)
TypeError: not all arguments converted during string formatting
~~~

Converting a tuple data to `str` worked for me.

~~~
-        _logger.debug('bTemps: %s' % bTemps)
+        _logger.debug('bTemps: %s' % str(bTemps))
         tTemps = temps['T0']
-        _logger.debug('tTemps: %s' % tTemps)
+        _logger.debug('tTemps: %s' % str(tTemps))
~~~

### Dict key error

On creating `tempsDict`, I got this error.

~~~
Exception in thread Thread-35:
Traceback (most recent call last):
  File "/home/octoprint/.pyenv/versions/3.9.4/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/home/octoprint/.pyenv/versions/3.9.4/lib/python3.9/threading.py", line 1266, in run
    self.function(*self.args, **self.kwargs)
  File "/home/octoprint/.pyenv/versions/3.9.4/envs/octoprint/lib/python3.9/site-packages/octoprint_ThermalRunaway/__init__.py", line 135, in check_temps
    tempsDict['B']['current'] = bTemps[0]
KeyError: 'B'
~~~

Inserting placeholders for `'B'` and `'T0'` would be fine.

~~~
-        tempsDict = dict()
+        tempsDict = {'B' : {}, 'T0' : {}}
         tempsDict['B']['current'] = bTemps[0]
         # bCurrentTemp = bTemps[0]
         tempsDict['T0']['current'] = tTemps[0]
~~~

Here is all the diffs I made.

~~~
@@ -128,10 +127,10 @@
         delaysDict['T0'] = int(tDelayStr)

         bTemps = temps['B']
-        _logger.debug('bTemps: %s' % bTemps)
+        _logger.debug('bTemps: %s' % str(bTemps))
         tTemps = temps['T0']
-        _logger.debug('tTemps: %s' % tTemps)
+        _logger.debug('tTemps: %s' % str(tTemps))
-        tempsDict = dict()
+        tempsDict = {'B' : {}, 'T0' : {}}
         tempsDict['B']['current'] = bTemps[0]
         # bCurrentTemp = bTemps[0]
         tempsDict['T0']['current'] = tTemps[0]
@@ -307,7 +306,7 @@
     # Temperatures hook
     def get_temps(self, comm, parsed_temps):
         _logger.debug('Temps received')  # Log that temps have been received
-        temps = parsed_temps
+        temps = parsed_temps.copy()
         if temps == parsed_temps:
             _logger.debug('Spawning new thread...')  # Log that we are attempting to spawn a new thread to process the received temps in
             t = threading.Timer(0, self.check_temps, [temps])  # Create a threading Timer object
~~~

AlexVerrico, 2021-05-16:
G'Day @TheFlyingBadger and @zero-impedance.
I believe this issue is fixed by the latest release.
Cheers, Alex.

closed 2021-05-16 by AlexVerrico