AlexVerrico/Octoprint-ThermalRunaway #18: [Feature Request] Additional triggers during event

merrill-oakland, 2021-03-02:
**Is your feature request related to a problem? Please describe.**
Would like to improve response to possible thermal runaway event.

**Describe the solution you'd like**
Would like the ability to call an external script and/or system command when thermal runaway is detected. Currently, the response of sending M112 can do nothing if something like a MOSFET failure has occurred.
In my specific case I would want to curl to call an external URL that triggers a notification and kills power to the printer with an external switch.
Would prefer that it not require another plugin to intercept the gcode, as it's not clear if that would work reliably if there are comms issues with the printer.

**Describe alternatives you've considered**
GCODE System Commands

**Additional context**
N/A

AlexVerrico, 2021-03-03:
G'Day @merrill-oakland 
Sounds like a great idea, I'll look into implementing this in the coming weeks, in the meantime, PRs are always welcome 😉 
Cheers, Alex.

AlexVerrico, 2021-05-16:
G'Day @merrill-oakland 
Quick update on the status of this feature.
I've updated the core ThermalRunaway plugin to allow for "extension" plugins to add functionality like this.
I'll be looking into creating an extension to allow calling system commands in the coming weeks.
Cheers, Alex.

AlexVerrico, 2021-05-17:
G'Day @merrill-oakland 
I've knocked together a basic plugin which should allow you to do what you want. 
I probably won't add it to the official plugin repo for now, as I wouldn't call it "finished" yet, however it should work just fine.
[https://github.com/AlexVerrico/Octoprint-TR_SysCommand](https://github.com/AlexVerrico/Octoprint-TR_SysCommand)
Cheers, Alex.

closed 2021-05-17 by AlexVerrico