AlexVerrico/Octoprint-ThermalRunaway #22: M112 sent some seconds after print job started

StephanBeutel, 2021-05-16:
**Describe the bug**
Printer gets M112 shutdown command some seconds after print job started

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print job

**Expected behavior**
Print job will be performed

**Screenshots**
Output from Octoprint.log:

2021-05-16 20:21:40,905 - octoprint.plugins.detailedprogress - INFO - Printing started. Detailed progress started.
2021-05-16 20:21:40,948 - octoprint.plugins.action_command_notification - INFO - Got a notification: Bed Heating...
2021-05-16 20:21:41,952 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:42,600 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:42,944 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2021-05-16 20:21:42,952 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:43,953 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:44,601 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:44,952 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:45,953 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:46,601 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:46,953 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:47,954 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:48,607 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:48,953 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:49,954 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:50,601 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:50,912 - octoprint.plugins.detailedprogress - INFO - Message: Good accuracy
2021-05-16 20:21:50,953 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:51,954 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:52,601 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:52,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:53,961 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:54,602 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:54,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:55,961 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:56,602 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:56,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:57,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:58,600 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:58,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:21:59,960 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:22:00,602 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:22:00,915 - octoprint.plugins.detailedprogress - INFO - Message: Good accuracy
2021-05-16 20:22:00,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:22:01,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:22:02,602 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:22:02,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount == 1 for B
2021-05-16 20:22:03,958 - octoprint.plugins.ThermalRunaway - WARNING - thermalLowCount > 2 for B
2021-05-16 20:22:03,959 - octoprint.util.comm - INFO - Force-sending M112 to the printer
2021-05-16 20:22:03,992 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline after error"
2021-05-16 20:22:04,002 - octoprint.plugins.ThermalRunaway - CRITICAL - Thermal Runaway (under temp) caught on heater B. Reported temp is 34.17, set temp is 60.0 

**Environment information:**
 - Octoprint Version: 1.6.1
 - Browser: Chrome
 - Python Version: 3.7.3
 - Operating system that Octoprint is running on: OctoPi 0.18.0

**Additional context**
Mega Zero V2.0 with SKR mini mz, BLTouch and BTT Smart Filament Sensor

AlexVerrico, 2021-05-17:
G'Day @StephanBeutel 
Can you confirm that the bed was approaching the target temperature?
If so, it may be caused by your bed heating too slowly. You may need to increase the bed delay in the settings:
[image]
Cheers, Alex.

StephanBeutel, 2021-05-17:
Hi, not as the M112 was sent before the bed was able to reach the temp. The bed takes something between 1-2 minutes to heat up.
Print started at 20:21:40 and M112 was sent at 20:22:03. Had no problems before upgrading t the actual version and did not touch the settings.

AlexVerrico, 2021-05-17:
interesting, have you tried again with the latest version that I released this morning?

ericbsmith42, 2021-05-17:
I started having this same issue this morning when I upgraded to v3.0.1. Basically as soon as I set a temperature the thermal runaway plugin instantly kills my Ender 5 printer, disconnecting. If I set an absurdly long time in the Max Bed Time (say 600 seconds) then there's no shutdown/disconnect.

AlexVerrico, 2021-05-17:
Hmm, i'll have to try and replicate this for further testing. In the meantime, you can install the previous version from here: https://github.com/AlexVerrico/Octoprint-ThermalRunaway/archive/refs/tags/2.0.1.zip

Ukchris79, 2021-05-17:
I was having issues yesterday generally following a print during cooldown, i never thought about checking the log, but here is the output from the terminal, this only started to happen post update. 

Recv:  T:64.17 /0.00 B:51.78 /0.00 T0:64.17 /0.00 T1:26.53 /0.00 @:0 B@:0 @0:0 @1:0
Recv:  T:63.52 /0.00 B:51.63 /0.00 T0:63.52 /0.00 T1:26.53 /0.00 @:0 B@:0 @0:0 @1:0
Recv:  T:62.78 /0.00 B:51.63 /0.00 T0:62.78 /0.00 T1:26.65 /0.00 @:0 B@:0 @0:0 @1:0
Send: M112
Send: N14076 M11237
Send: N14077 M104 T0 S036
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error"

AlexVerrico, 2021-05-18:
G'Day all.
I have not been successful in reproducing this behavior for testing, however I believe that I have found the cause of the issue.
I have created a new branch with a new configuration option which should fix this issue (by default it won't be fixed, you will need to open the settings page and uncheck the option at the very bottom.)
You can install it using this link: https://github.com/AlexVerrico/Octoprint-ThermalRunaway/archive/refs/heads/bug-22.zip
If this changes anything (or if it doesn't) please let me know.
Cheers, Alex.

Jon-So, 2021-05-18:
Hey Alex, I tried the bug fix as well but the printers all still HaLT reset...

AlexVerrico, 2021-05-18:
@Jon-So 
did you change this setting to look like this?
[image]

dwery, 2021-05-18:
Same thing here, happened after the last update. Was working nicely before.

Jon-So, 2021-05-18:
Hey Alex, Yes I unchecked it and everything started working again... thanks for the fix... I almost took the printer apart thinking the mosfet had died.  lol

AlexVerrico, 2021-05-18:
Thanks for the update @Jon-So, glad to hear it's working for you
@dwery Have you tried installing the plugin from this link:
https://github.com/AlexVerrico/Octoprint-ThermalRunaway/archive/refs/heads/bug-22.zip
And then changing the setting as described above?

Jon-So, 2021-05-18:
yes I used the bug fix file and I unchecked it. everything was working when it was unchecked...

AlexVerrico, 2021-05-18:
Interesting, the change was made because I got multiple reports of the plugin failing to catch real thermal runaways with the previous version, but clearly the changes don't suit all setups. If I can get confirmation from at least one more person that the new settings option fixes this problem then I'll release a new version containing the fix.

Ano10, 2021-05-18:
Hi,
Since update 3.0.1 I started to have issue as well. Using your fix make your plugin work like before updating.
Thanks for your work.

AlexVerrico, 2021-05-18:
Thanks for letting me know @Ano10

AlexVerrico, 2021-05-18:
I have now merged the fix into the main branch and released a new version, you may need to force a check for updates to install the new version.

AlexVerrico, 2022-01-14:
Closing due to inactivity

closed 2022-01-14 by AlexVerrico