AlexVerrico/Octoprint-ThermalRunaway #25: [BUG] M112 sent while printer/bed is cooling down

bcorcoran, 2021-05-18:
Prusa MK3S on 3.10.0 firmware (but this occurred on 3.8.1 as well, so doesnt seem FW related)

Octoprint debug log mentions:

```
2021-05-18 15:20:33,041 - octoprint.plugins.ThermalRunaway - WARNING - thermalHighCount > 2 for T0
2021-05-18 15:20:33,042 - octoprint.util.comm - INFO - Force-sending M112 to the printer
```

But the printer was cooling down, observe:

```
Recv: T:151.8 /0.0 B:49.8 /0.0 T0:151.8 /0.0 @:0 B@:0 P:43.6 A:32.1
Recv: T:150.8 /0.0 B:49.5 /0.0 T0:150.8 /0.0 @:0 B@:0 P:33.1 A:32.1
Recv: T:149.9 /0.0 B:49.5 /0.0 T0:149.9 /0.0 @:0 B@:0 P:32.1 A:32.2
Recv: T:149.2 /0.0 B:49.2 /0.0 T0:149.2 /0.0 @:0 B@:0 P:32.0 A:32.0
Recv: T:148.4 /0.0 B:48.9 /0.0 T0:148.4 /0.0 @:0 B@:0 P:32.0 A:32.0
Recv: T:147.7 /0.0 B:49.0 /0.0 T0:147.7 /0.0 @:0 B@:0 P:31.9 A:31.9
Recv: T:146.9 /0.0 B:48.9 /0.0 T0:146.9 /0.0 @:0 B@:0 P:31.9 A:32.2
Recv: T:146.0 /0.0 B:48.8 /0.0 T0:146.0 /0.0 @:0 B@:0 P:31.9 A:32.0
Recv: T:145.3 /0.0 B:48.4 /0.0 T0:145.3 /0.0 @:0 B@:0 P:31.9 A:32.0
Recv: T:144.5 /0.0 B:48.2 /0.0 T0:144.5 /0.0 @:0 B@:0 P:31.9 A:32.2
Recv: T:143.6 /0.0 B:48.3 /0.0 T0:143.6 /0.0 @:0 B@:0 P:31.9 A:31.9
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Recv: KILL: 2
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error"
Send: M112
Closing serial port due to emergency stop M112.
Send: N5 M112*36
Send: N6 M104 T0 S0*39
Send: N7 M140 S0*98
Connection closed, closing down monitor
Changing monitoring state from "Error" to "Offline after error"
```

My thermistor is fine, and the printer/octoprint operate normally without the plugin installed.

bcorcoran, 2021-05-18:
Seems related to #5 but this happens after a print job completes as well.

For now, I don't have the plugin installed as it's too disruptive.

AlexVerrico, 2021-05-18:
G'Day @bcorcoran 
Please try installing a patched version of this plugin as described here: https://github.com/AlexVerrico/Octoprint-ThermalRunaway/issues/22#issuecomment-842723546
And then let me know how you go.
Cheers, Alex.

bcorcoran, 2021-05-18:
Will do, but may be a few days as I have several prints queued up

AlexVerrico, 2021-05-18:
No worries

AlexVerrico, 2021-05-18:
I have now merged the fix into the main branch and released a new version, you may need to force a check for updates to install the new version.