AlexVerrico/Octoprint-ThermalRunaway #32: [BUG] M112 is sent due to a warning a day before

florianwgnr, 2022-02-19:
**Describe the bug**
M112 is sent due to a warning a day before. 

**What happened**
When i start my printer together with the raspberry pi, the printer sometimes stops after 20 seconds with M112. Octoprints log shows the following:

**Log-File**
###Log 2022-02-16
2022-02-16 21:55:44,118 - octoprint.plugins.ThermalRunaway - WARNING - B temperature is now too low, (low=20.62, current=20.62, set=64.0)
2022-02-16 21:55:44,118 - octoprint.plugins.ThermalRunaway - WARNING - T0 temperature is now too low, (low=20.62, current=20.62, set=120.0)
2022-02-16 21:55:44,386 - octoprint.plugins.continuousprint - INFO - Initialized
2022-02-16 21:55:44,387 - octoprint.plugins.continuousprint - INFO - Retry on pause: False (max_retries 3, threshold 3600s)
2022-02-16 21:55:44,388 - octoprint.plugins.continuousprint - INFO - Continuous Print Plugin started
2022-02-16 21:55:44,388 - octoprint.plugins.detailedprogress - INFO - OctoPrint-DetailedProgress loaded!
2022-02-16 21:55:44,388 - octoprint.plugins.draggable_files - INFO - Hello World!
2022-02-16 21:55:44,389 - octoprint.plugins.stats - INFO - Printer Stats
2022-02-16 21:55:44,471 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-02-16 21:55:44,846 - octoprint.plugins.toptemp - INFO - TopTemp is initialized
2022-02-16 21:55:45,258 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://octopi.local/', 'path': '/', 'query_string': 'l10n=de'}
2022-02-16 21:55:46,095 - octoprint.plugins.ThermalRunaway - WARNING - **B temperature has been too low** for 1.9770498275756836 seconds, (low=20.62, current=20.62, set=64.0)
2022-02-16 21:55:46,096 - octoprint.plugins.ThermalRunaway - INFO - T0 temperature is no longer too low after 1.9771485328674316 seconds, (low=20.62, current=21.87, set=120.0)

###Log 2022-02-17
2022-02-17 16:31:55,480 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-02-17 16:31:56,656 - octoprint.plugins.ThermalRunaway - **WARNING - B temperature has been too low for 66972.53778862953** seconds, (low=20.62, current=20.62, set=64.0)
2022-02-17 16:31:56,657 - octoprint.util.comm - INFO - Force-sending M112 to the printer

**Note**

M112 is sent because B-temp was low for 66972 seconds (13 h). So something is not cleared before shutdown. 

**Environment information:**
 - Octoprint Version: 0.17.0
 - Python Version: Python 3.7.3
 - Plugin Version: Thermal-Runaway (3.1.0) 
 - Operating system that Octoprint is running on: Raspbian GNU/Linux 10 (buster)

BobVul, 2022-03-12:
Hmm, in this case do you keep the Pi running while restarting the printer? If you're also restarting the Pi, it should not be able to track warnings through that.

At the moment, the temperature/warning tracking is only reset when the set temperature changes. If your printer is set to heat very soon after connecting (and was last in a warning state), that'd be why it's triggering here.

I suppose the best way to fix this would be to reset tracking on machine connect. I'll do a PR when I have time.

BobVul, 2022-03-16:
> The Pi is also restarted and configured to heat up soon after connecting.

Could you confirm that also logs a `octoprint.startup - INFO - Starting OctoPrint` in between the warning events?

I'm still confused how this can be happening, as I don't think this plugin ever saves anything to any form of persistent storage - so a Pi restart (reboot, not suspend/resume) should clear everything.

AlexVerrico, 2022-04-13:
I don't have time to look into this fully right now, but BobVul is correct, the plugin has no persistent storage for warnings (meaning that restarting the Pi will clear all warnings).

florianwgnr, 2022-04-13:
> Could you confirm that also logs a `octoprint.startup - INFO - Starting OctoPrint` in between the warning events?

I´m sorry but i deleted the file. What I do know is that the Pi was definitely offline because it´s power-supply is connected to the printer and both are off overnight. 

Another hint is that my OctoPrint-Telegram-Bot sent me two messages with "hello i´m online" (meaning octoprint was started successfully), one at 16:31 (2022-02-17 16:31, see log) and another one ten minutes later when i restarted the Pi and the printer. 

Maybe it is linked to my start routine: OctoPrint is configured to send the following GCode to my printer after establishing a connection:

`M106 P0 S128 ;Set FAN to 50% (Board and Hotend)
M140 S64 ;Heat up bed
M104 S120 ;Heat up extruder to low standby`

Maybe this is executed just a moment before the plugin starts, the bed is still cold and therefore it fails?